mvukov/rules_ros2 #1: No msg shown in listener logging.

EvanLiUAV, 2021-06-07:
Talker log:
[INFO] [1623070813.542118087] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [1623070814.042105862] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [1623070814.542064991] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [1623070815.042087247] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1623070815.542077403] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1623070816.042034178] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1623070816.542057880] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1623070817.042046168] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1623070817.542036208] [minimal_publisher]: Publishing: 'Hello, world! 8'
[INFO] [1623070818.042026548] [minimal_publisher]: Publishing: 'Hello, world! 9'
Listener log:
[INFO] [1623070819.042303829] [minimal_subscriber]: I heard: ''
[INFO] [1623070819.542297902] [minimal_subscriber]: I heard: ''
[INFO] [1623070820.042197016] [minimal_subscriber]: I heard: ''
[INFO] [1623070820.542269727] [minimal_subscriber]: I heard: ''

Got msg when talker is publish , but the msgs is empty.

EvanLiUAV, 2021-06-07:
Is there any configuration needed by cyclone dds?

mvukov, 2021-06-07:
Hi, thanks for testing the code. Yeah, I have the same issue on my machine and I opened a thread on ROS Answers, see https://answers.ros.org/question/379673/problems-with-a-bazel-build-of-ros2/.

I (double-) checked compilation flags for ported packages, that should be fine. But I suspect I missed something critical. It can also be that there is some silent failure within ROS2 stack -- just thinking out loud.

I also checked whether there is a custom config for CycloneDDS in a colcon-based build: there is none.

mvukov, 2021-06-07:
Hi, thanks for testing out the code. I have the same issue on my machine, that's why I opened a thread on ROS answers, see https://answers.ros.org/question/379673/problems-with-a-bazel-build-of-ros2/.

I (double-) checked compilation flags for ported packages, that should be fine. But, I suspect I missed something critical as CMake build is quite involved. It can also be that there is some silent failure (resulting in no visible warning/error logs) in the ROS2 stack  -- just thinking out loud.

I also checked whether there is CycloneDDS specific config in a regular colcon-based build: there is none AFAICS.

EvanLiUAV, 2021-06-08:
I can see ros2 topic info /topic
Topic: /topic
Publisher count: 1
Subscriber count: 1

EvanLiUAV, 2021-06-08:
> Hi, thanks for testing out the code. I have the same issue on my machine, that's why I opened a thread on ROS answers, see [...]

Thanks for the reply,  maybe we can check if the generated message is correct because the topic echo shown that the message is not valid.
``` bash 
evan@evan:~$ ros2 topic echo /topic
The passed message type is invalid
```

mvukov, 2021-06-08:
Hi Evan, thanks for the additional info. I can make a more thorough comparison of the generated code w.r.t. a CMake build. It can be indeed that I missed something there.

EvanLiUAV, 2021-06-08:
> Hi Evan, thanks for the additional info. I can make a more thorough comparison of the generated code w.r.t. a CMake build. It can be indeed that I missed something there.
I found something different between Bazel generate code and original code generate by CMAKE command 

original cpp generated by CMAKE
``` cpp
// generated from rosidl_typesupport_cpp/resource/idl__type_support.cpp.em
// with input from std_msgs:msg/String.idl
// generated code does not contain a copyright notice

#include "cstddef"
#include "rosidl_runtime_c/message_type_support_struct.h"
#include "std_msgs/msg/detail/string__struct.hpp"
#include "rosidl_typesupport_cpp/identifier.hpp"
#include "rosidl_typesupport_cpp/message_type_support.hpp"
#include "rosidl_typesupport_c/type_support_map.h"
#include "rosidl_typesupport_cpp/message_type_support_dispatch.hpp"
#include "rosidl_typesupport_cpp/visibility_control.h"
#include "rosidl_typesupport_interface/macros.h"

namespace std_msgs
{

namespace msg
{

namespace rosidl_typesupport_cpp
{

typedef struct _String_type_support_ids_t
{
  const char * typesupport_identifier[2];
} _String_type_support_ids_t;

static const _String_type_support_ids_t _String_message_typesupport_ids = {
  {
    "rosidl_typesupport_fastrtps_cpp",  // ::rosidl_typesupport_fastrtps_cpp::typesupport_identifier,
    "rosidl_typesupport_introspection_cpp",  // ::rosidl_typesupport_introspection_cpp::typesupport_identifier,
  }
};

typedef struct _String_type_support_symbol_names_t
{
  const char * symbol_name[2];
} _String_type_support_symbol_names_t;

#define STRINGIFY_(s) #s
#define STRINGIFY(s) STRINGIFY_(s)

static const _String_type_support_symbol_names_t _String_message_typesupport_symbol_names = {
  {
    STRINGIFY(ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_fastrtps_cpp, std_msgs, msg, String)),
    STRINGIFY(ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_introspection_cpp, std_msgs, msg, String)),
  }
};

typedef struct _String_type_support_data_t
{
  void * data[2];
} _String_type_support_data_t;

static _String_type_support_data_t _String_message_typesupport_data = {
  {
    0,  // will store the shared library later
    0,  // will store the shared library later
  }
};

static const type_support_map_t _String_message_typesupport_map = {
  2,
  "std_msgs",
[... 148 lines cut ...]
<   "std_msgs",
<   &_String_message_typesupport_ids.typesupport_identifier[0],
<   &_String_message_typesupport_symbol_names.symbol_name[0],
<   &_String_message_typesupport_data.data[0],
< };
< 
< static const rosidl_message_type_support_t String_message_type_support_handle = {
<   ::rosidl_typesupport_cpp::typesupport_identifier,
<   reinterpret_cast<const type_support_map_t *>(&_String_message_typesupport_map),
<   ::rosidl_typesupport_cpp::get_message_typesupport_handle_function,
< };
< 
< }  // namespace rosidl_typesupport_cpp
< 
< }  // namespace msg
< 
< }  // namespace std_msgs
---
> #include "std_msgs/msg/detail/string__rosidl_typesupport_introspection_cpp.hpp"
91c20
<   return &::std_msgs::msg::rosidl_typesupport_cpp::String_message_type_support_handle;
---
>   return ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME(rosidl_typesupport_introspection_cpp, std_msgs, msg, String)();

```

mvukov, 2021-06-08:
OK, thanks. Just a quick question: for a colon-based build you used ROS2 foxy as well, right?

BTW, to make this apples-to-apples comparison, you should remove fastrtps from your CMake build. I will do that exercise as well.

EvanLiUAV, 2021-06-08:
> OK, thanks. Just a quick question: for a colon-based build you used ROS2 foxy as well, right?
> 
> BTW, to make this apples-to-apples comparison, you should remove fastrtps from your CMake build. I will do that exercise as well.

I fix this bug by modified rmw_cyberdds package. the patch is as below based on foxy branch:
``` cpp
diff --git a/rmw_cyclonedds_cpp/src/serdata.cpp b/rmw_cyclonedds_cpp/src/serdata.cpp
index 05b3286..b3382cc 100644
--- a/rmw_cyclonedds_cpp/src/serdata.cpp
+++ b/rmw_cyclonedds_cpp/src/serdata.cpp
@@ -60,14 +60,16 @@ using ResponseTypeSupport_cpp = rmw_cyclonedds_cpp::ResponseTypeSupport<
   rosidl_typesupport_introspection_cpp::ServiceMembers,
   rosidl_typesupport_introspection_cpp::MessageMembers>;
 
-static bool using_introspection_c_typesupport(const char * typesupport_identifier)
-{
-  return typesupport_identifier == rosidl_typesupport_introspection_c__identifier;
+static bool using_introspection_c_typesupport(
+    const char* typesupport_identifier) {
+  return !std::string(typesupport_identifier)
+      .compare(rosidl_typesupport_introspection_c__identifier);
 }
 
-static bool using_introspection_cpp_typesupport(const char * typesupport_identifier)
-{
-  return typesupport_identifier == rosidl_typesupport_introspection_cpp::typesupport_identifier;
+static bool using_introspection_cpp_typesupport(
+    const char* typesupport_identifier) {
+  return !std::string(typesupport_identifier)
+      .compare(rosidl_typesupport_introspection_cpp::typesupport_identifier);
 }
 
 void * create_message_type_support(
```

mvukov, 2021-06-09:
Closed via #2 .

closed 2021-06-09 by mvukov