mvukov/rules_ros2 #4: Failed to load shared library

Sanjaer, 2022-05-06:
Hi, I love the idea of this project. I've been trying to use it but when everything is built I get the following error when trying to run the examples:
```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl init options: failed to load shared library 'external/ros2_rmw_cyclonedds/librmw_cyclonedds.so' due to LoadLibrary error: external/ros2_rmw_cyclonedds/librmw_cyclonedds.so: cannot open shared object file: No such file or directory, at external/ros2_rcutils/src/shared_library.c:84, at external/ros2_rmw_implementation/rmw_implementation/src/functions.cpp:90, at external/ros2_rcl/rcl/src/rcl/init_options.c:61
Aborted (core dumped)
```
Is there any documentation where I can check the proper usage of this rules for bazel?
Thanks in advance

mvukov, 2022-05-06:
Hi Pablo,

Thanks! It looks to me that you're trying to run the example nodes without `bazel run` cmd. This is possible, but is a bit more involved -- and not that practical IMHO. I've put together a small [readme](https://github.com/mvukov/rules_ros2/blob/main/examples/chatter/README.md) .

If you really want to run the binaries w/o Bazel then you need to be in a correct runtime folder, since the nodes load rmw at runtime. For talker that would be:
```sh
cd bazel-bin/examples/chatter/talker.runfiles/com_github_mvukov_rules_ros2/
./examples/chatter/talker
```
(not very practical...)

Indeed, there are no docs yet for the code in this repo -- the current state is more/less the 1st stab I have done to get nodes talking and working C/C++ code-generation for interfaces. From the user perspective, the two examples depict what you can do today with this repo. The other example illustrates how to create custom interfaces. The approach is: define ros2_interface_library target with your msgs/srvs/actions, then use that in a cpp_ros2_interface_library target that you can use as a dependency in a C++ target.

Let me know if you need more help.

Sanjaer, 2022-05-09:
Hi, thanks for your answer. Now it works! I'll keep working and I'll ask you if I encounter more difficulties. Thanks!

closed 2022-05-09 by Sanjaer