mvukov/rules_ros2 #7: Running --spawn_strategy=local

Sanjaer, 2022-05-19:
Hi, for validation purposes I need to run this repo with --spawn_strategy=local. However it breaks the dependencies:
```

$ bazelisk run --spawn_strategy=local //examples/chatter:talker
Starting local Bazel server and connecting to it...
INFO: Analyzed target //examples/chatter:talker (64 packages loaded, 1703 targets configured).
INFO: Found 1 target...
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rmw_cyclonedds/BUILD.bazel:4:10: Compiling rmw_cyclonedds_cpp/src/rmw_node.cpp failed: undeclared inclusion(s) in rule '@ros2_rmw_cyclonedds//:rmw_cyclonedds':
this rule is missing dependency declarations for the following files included by 'rmw_cyclonedds_cpp/src/rmw_node.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/gid.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__builder.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__traits.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/node_entities_info.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__builder.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__traits.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/participant_entities_info.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__builder.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__traits.hpp'
Target //examples/chatter:talker failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 17.207s, Critical Path: 4.56s
INFO: 39 processes: 4 internal, 35 local.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully

```

Any idea on why this might happen and how can it be fixed?

mvukov, 2022-05-19:
I've never used this option. Why do u need it?

Off the top of my head it could be that some deps are not directly enforced and that Bazel complains about that. Is this the only error? What happens with `--keep_going`?

Sanjaer, 2022-05-20:
We are running py_test() in order to validate our app, which needs access to elements outside the sandbox. By using that option, sandboxing is disabled and gets everything directly built on the output tree. A single bazelisk test --spawn_strategy=local test will build the application and in turn the ROS libraries, and the spawn_strategy=local option will be applied to all steps of the process, since ROS is a dependency of the app binary, which is a dependency of the test.

By running with --keep_going, it simply prints more dependency errors on the interfaces:

```
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rmw_dds_common/BUILD.bazel:17:27: Compiling external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/node_entities_info__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rmw_dds_common//:cpp_rmw_dds_common':
this rule is missing dependency declarations for the following files included by 'external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/node_entities_info__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__rosidl_typesupport_introspection_cpp.hpp'
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rmw_dds_common/BUILD.bazel:17:27: Compiling external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/detail/participant_entities_info__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rmw_dds_common//:cpp_rmw_dds_common':
this rule is missing dependency declarations for the following files included by 'external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/detail/participant_entities_info__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/node_entities_info__struct.hpp'
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rmw_dds_common/BUILD.bazel:17:27: Compiling external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/gid__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rmw_dds_common//:cpp_rmw_dds_common':
this rule is missing dependency declarations for the following files included by 'external/ros2_rmw_dds_common/cpp_rmw_dds_common_cpp/rmw_dds_common/msg/gid__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__rosidl_typesupport_introspection_cpp.hpp'

.....
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rcl_interfaces/BUILD.bazel:90:27: Compiling external/ros2_rcl_interfaces/cpp_statistics_msgs_cpp/statistics_msgs/msg/metrics_message__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rcl_interfaces//:cpp_statistics_msgs':
this rule is missing dependency declarations for the following files included by 'external/ros2_rcl_interfaces/cpp_statistics_msgs_cpp/statistics_msgs/msg/metrics_message__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/detail/metrics_message__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/builtin_interfaces/msg/detail/time__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/detail/statistic_data_point__struct.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/detail/metrics_message__rosidl_typesupport_introspection_cpp.hpp'
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rcl_interfaces/BUILD.bazel:84:27: Compiling external/ros2_rcl_interfaces/cpp_rosgraph_msgs_cpp/builtin_interfaces/msg/detail/duration__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rcl_interfaces//:cpp_rosgraph_msgs':
this rule is missing dependency declarations for the following files included by 'external/ros2_rcl_interfaces/cpp_rosgraph_msgs_cpp/builtin_interfaces/msg/detail/duration__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/builtin_interfaces/msg/detail/duration__struct.hpp'
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/c9f305f88dd2169115dab55cd07eb97d/external/ros2_rcl_interfaces/BUILD.bazel:84:27: Compiling external/ros2_rcl_interfaces/cpp_rosgraph_msgs_cpp/builtin_interfaces/msg/time__type_support.cpp failed: undeclared inclusion(s) in rule '@ros2_rcl_interfaces//:cpp_rosgraph_msgs':
this rule is missing dependency declarations for the following files included by 'external/ros2_rcl_interfaces/cpp_rosgraph_msgs_cpp/builtin_interfaces/msg/time__type_support.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/builtin_interfaces/msg/detail/time__struct.hpp'

```
...and keeps on going this way

mvukov, 2022-05-20:
OK, this is interesting: if I just run `bazel run --spawn_strategy=local //examples/chatter:talker` this works fine for me locally. Did you run this command after cleaning the Bazel cache?

Sanjaer, 2022-05-23:
That's weird, this is from a fresh clone, trying to run talker just like you:
```
 bazelisk run --spawn_strategy=local //examples/chatter:talker
Starting local Bazel server and connecting to it...
DEBUG: Rule 'ros2_rosidl_typesupport' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921586 +0200"
DEBUG: Repository ros2_rosidl_typesupport instantiated at:
  /home/mobuser/pruebas/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/mobuser/pruebas/rules_ros2/repositories/repositories.bzl:173:10: in ros2_repositories
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rcpputils' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921172 +0200"
DEBUG: Repository ros2_rcpputils instantiated at:
  /home/mobuser/pruebas/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/mobuser/pruebas/rules_ros2/repositories/repositories.bzl:69:10: in ros2_repositories
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rosidl' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622892676 +0200"
DEBUG: Repository ros2_rosidl instantiated at:
  /home/mobuser/pruebas/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/mobuser/pruebas/rules_ros2/repositories/repositories.bzl:44:10: in ros2_repositories
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rmw_implementation' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921874 +0200"
DEBUG: Repository ros2_rmw_implementation instantiated at:
  /home/mobuser/pruebas/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/mobuser/pruebas/rules_ros2/repositories/repositories.bzl:85:10: in ros2_repositories
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
INFO: Analyzed target //examples/chatter:talker (64 packages loaded, 1703 targets configured).
INFO: Found 1 target...
INFO: From Action external/ros2_rmw_dds_common/rmw_dds_common/rosidl_adapter_map.idls:
Reading input file: external/ros2_rmw_dds_common/rmw_dds_common/msg/Gid.msg
Writing output file: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/Gid.idl
Reading input file: external/ros2_rmw_dds_common/rmw_dds_common/msg/NodeEntitiesInfo.msg
Writing output file: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/NodeEntitiesInfo.idl
Reading input file: external/ros2_rmw_dds_common/rmw_dds_common/msg/ParticipantEntitiesInfo.msg
Writing output file: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/ParticipantEntitiesInfo.idl
INFO: From Action external/ros2_rcl_interfaces/builtin_interfaces/rosidl_adapter_map.idls:
Reading input file: external/ros2_rcl_interfaces/builtin_interfaces/msg/Duration.msg
Writing output file: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/builtin_interfaces/msg/Duration.idl
Reading input file: external/ros2_rcl_interfaces/builtin_interfaces/msg/Time.msg
Writing output file: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/builtin_interfaces/msg/Time.idl
ERROR: /home/mobuser/.cache/bazel/_bazel_mobuser/92a89763c8c912c5450ef8e34cc34689/external/ros2_rmw_cyclonedds/BUILD.bazel:4:10: Compiling rmw_cyclonedds_cpp/src/rmw_node.cpp failed: undeclared inclusion(s) in rule '@ros2_rmw_cyclonedds//:rmw_cyclonedds':
this rule is missing dependency declarations for the following files included by 'rmw_cyclonedds_cpp/src/rmw_node.cpp':
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/gid.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/gid__struct.hpp'
[... cut ...]
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__builder.hpp'
  'bazel-out/k8-fastbuild/bin/external/ros2_rmw_dds_common/rmw_dds_common/msg/detail/participant_entities_info__traits.hpp'
Target //examples/chatter:talker failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 34.771s, Critical Path: 5.62s
INFO: 121 processes: 50 internal, 71 local.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
```

mvukov, 2022-07-28:
I use Bazel 5.1.1. All stuff I use locally is committed to this repo. @Sanjaer Did you solve this in the meantime?

Sanjaer, 2022-07-30:
We did a work around, as it is for tests, we build it with
spawn-strategy=sanboxed and use the binary to build the tests as an
external dependency.
I'll check my version on Monday just for the sake of comparison.

Sanjaer, 2022-08-10:
Hi, I completely forgot, we use bazel 5.2.0 (in fact we use bazelisk)

mvukov, 2022-10-10:
Just FYI, check `ros2_test` macro for testing. BTW, dunno how to help you with your issue. Can you make a minimal Dockerfile that reproduces your issues?

Sanjaer, 2022-10-11:
Hi mvukov. Unfortunately I'm not working on that project anymore and I
don't have access to the code. However we found a workaround as I told you
before so maybe we can close this issue :)

closed 2022-10-12 by mvukov