mvukov/rules_ros2 #8: [feature request] README file

kev-the-dev, 2022-06-04:
This looks like an awesome project! If I find some time I'd love to help get more of ros2 running within bazel! 

Wondering if you'd consider adding a brief README. Some things that may be helpful:

* A list of which things are supported or not supported currently
* Any expectations you have for contributions, like style or format of PRs
* Maybe a quick demo of the project, or a reference to the example already present in the repo and how one could reproduce it (like a command)

mvukov, 2022-06-05:
OK, OK, I'll put together a readme, finally :)

In the meantime, you can take a look at `examples/chatter`. There is a short readme in that folder. Afterward, you can take a look at `examples/actions`.

mvukov, 2022-06-26:
[kev-the-dev](https://github.com/kev-the-dev) I put together a short readme. Please let me know if you would like to see therein something more.

kev-the-dev, 2022-06-26:
This is great! Thank you

closed 2022-06-26 by kev-the-dev