mvukov/rules_ros2 #9: [question] ROS2 Nodes

paulsammut, 2022-07-25:
Hello, thanks so much for making this project! I have a question:

How did you figure out the minimum set of nodes/packages required to "bazelify" Foxy? 

I'm trying to do a similar project, except with Buck and with ROS2 Humble, and have been making use of the colcon graph output to figure out the dependency graph, but I'm unsure of which packages I can get rid of.

Thanks for any help!
#question

ahans, 2022-07-26:
I had a similar question [here](https://answers.ros.org/question/398300/how-to-find-dependencies-on-shared-objects-in-ros2/). Instead of doing everything in Bazel, my starting point is a "fat archive" distribution. I'm using a Python script to parse the tree and use the information in `package.xml` files (I think that's the only thing colcon relies on), but also use some CMake files. A proper answer to my question over there is still pending. The only thing I got was a reference to this very project!

To find out the minimal set of packages, I would start from a package I want to use as a starting point, e.g., rclcpp, and then walk the dependency tree backwards from there. In my experience this still requires some manual adjustments due to system dependencies assumed to be present and what not, but it's definitely better than a full manual effort.

mvukov, 2022-07-26:
This was a while ago, but I think I was lazy to create a script to visualize a dependency graph so I was manually inspecting package.xml files, starting from rclc and when I got that up-n-running I investigated how to build rclcpp. There was a bit of trial-n-error there to figure out what was strictly needed for compilation. Creating a script to inspect deps and create a graph shouldn't be that hard. For my rules_ros repo I created a Python script that used catkin for package discovery.

What you can do is to create a dependency graph for e.g. rclcpp using Bazel and this repo:
```sh
bazel query "kind(cc_library, deps(@ros2_rclcpp//:rclcpp))" --output=graph --noimplicit_deps
```
And then you can visualize the output e.g. [here](https://dreampuf.github.io/GraphvizOnline/#digraph%20mygraph%20%7B%0A%20%20node%20%5Bshape%3Dbox%5D%3B%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_libstatistics_collector%2F%2F%3Alibstatistics_collector%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcl%2F%2F%3Arcl%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcl%2F%2F%3Arcl_yaml_param_parser%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_statistics_msgs%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcpputils%2F%2F%3Arcpputils%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rcutils%2F%2F%3Arcutils%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rmw%2F%2F%3Armw%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rmw%2F%2F%3Armw_cpp%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_cpp%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rosidl_typesupport%2F%2F%3Arosidl_typesupport_c%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_rosidl_typesupport%2F%2F%3Arosidl_typesupport_cpp%22%0A%20%20%22%40ros2_rclcpp%2F%2F%3Arclcpp%22%20-%3E%20%22%40ros2_tracing%2F%2F%3Atracetools%22%0A%20%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%0A%20%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_cpp%22%0A%20%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_typesupport_introspection_c%22%0A%20%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_typesupport_introspection_cpp%22%0A%20%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_rcl_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_builtin_interfaces%5Cn%40ros2_rcl_interfaces%2F%2F%3Acpp_rosgraph_msgs%22%20-%3E%20%22%40ros2_rosidl_typesupport%2F%2F%3Arosidl_typesupport_cpp%22%0A%20%20%22%40ros2_libstatistics_collector%2F%2F%3Alibstatistics_collector%22%0A%20%20%22%40ros2_libstatistics_collector%2F%2F%3Alibstatistics_collector%22%20-%3E%20%22%40ros2_rcl%2F%2F%3Arcl%22%0A%20%20%22%40ros2_libstatistics_collector%2F%2F%3Alibstatistics_collector%22%20-%3E%20%22%40ros2_rcpputils%2F%2F%3Arcpputils%22%0A%20%20%22%40ros2_libstatistics_collector%2F%2F%3Alibstatistics_collector%22%20-%3E%20%22%40ros2_rcl_interfaces%2F%2F%3Acpp_statistics_msgs%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rcl%2F%2F%3Arcl_yaml_param_parser%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rcl_interfaces%2F%2F%3Ac_rcl_interfaces%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rcl_logging%2F%2F%3Arcl_logging_spdlog%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rcutils%2F%2F%3Arcutils%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_c%22%0A%20%20%22%40ros2_rcl%2F%2F%3Arcl%22%20-%3E%20%22%40ros2_tracing%2F%2F%3Atracetools%22%0A%20%20%22%40ros2_tracing%2F%2F%3Atracetools%22%0A%20%20%22%40ros2_tracing%2F%2F%3Atracetools%22%20-%3E%20%22%40ros2_tracing%2F%2F%3Atracetools_c%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40ros2_rcpputils%2F%2F%3Arcpputils%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40ros2_rcutils%2F%2F%3Arcutils%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40ros2_rmw%2F%2F%3Armw%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40cyclonedds%2F%2F%3Addsc%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40ros2_rmw%2F%2F%3Armw_cpp%22%0A%20%20%22%40ros2_rmw_implementation%2F%2F%3Armw_implementation%22%20-%3E%20%22%40ros2_rmw_dds_comm [... cut ...] Arosidl_runtime_c%22%0A%20%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_c%22%0A%20%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_c%22%20-%3E%20%22%40ros2_rosidl%2F%2F%3Arosidl_typesupport_interface%22%0A%20%20%22%40ros2_rosidl%2F%2F%3Arosidl_runtime_c%22%20-%3E%20%22%40ros2_rcutils%2F%2F%3Arcutils%22%0A%20%20%22%40ros2_rcutils%2F%2F%3Arcutils%22%0A%20%20%22%40ros2_rosidl%2F%2F%3Arosidl_typesupport_interface%22%0A%7D%0A).
This visualizes deps between the Bazel targets, should be a good starting point.

paulsammut, 2022-07-26:
Thanks a bunch for this. Did you find that the rclc was needed for rclcpp? Or did you just want to include both of them?

mvukov, 2022-07-28:
So, rclcpp is dependent on rcl and some C-based libraries but not rclc directly (my bad for claiming this earlier: I started the dependency analysis from rcl instead).

There are two examples in the `examples` folder. As there is no launch mechanism in place in this repo (yet), you have to start each node binary in a separate terminal. And this works just fine.