mvukov/rules_ros2 #14: unable to build any python examples

LiyouZhou, 2022-10-10:
commit : `106a7b615c6bcab0c96ea82d83c53351c6cb182e`
command: `bazel run //examples/chatter`
output: 
```
DEBUG: Rule 'ros2cli' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1663423078 +0200"
DEBUG: Repository ros2cli instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:331:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rosidl' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622892676 +0200"
DEBUG: Repository ros2_rosidl instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:45:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rosidl_typesupport' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921586 +0200"
DEBUG: Repository ros2_rosidl_typesupport instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:174:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rcpputils' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921172 +0200"
DEBUG: Repository ros2_rcpputils instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:70:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_rmw_implementation' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1622921874 +0200"
DEBUG: Repository ros2_rmw_implementation instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:86:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
DEBUG: Rule 'ros2_launch' indicated that a canonical reproducible form can be obtained by modifying arguments shallow_since = "1662324633 +0200"
DEBUG: Repository ros2_launch instantiated at:
  /home/liyou/work/rules_ros2/WORKSPACE:5:18: in <toplevel>
  /home/liyou/work/rules_ros2/repositories/repositories.bzl:306:10: in ros2_repositories
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/utils.bzl:233:18: in maybe
Repository rule new_git_repository defined at:
  /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/external/bazel_tools/tools/build_defs/repo/git.bzl:186:37: in <toplevel>
INFO: Analyzed target //examples/chatter:chatter (16 packages loaded, 1037 targets configured).
INFO: Found 1 target...
INFO: From Action external/ros2_rcl_interfaces/statistics_msgs/rosidl_adapter_map.idls:
Reading input file: external/ros2_rcl_interfaces/statistics_msgs/msg/MetricsMessage.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/665/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/MetricsMessage.idl
Reading input file: external/ros2_rcl_interfaces/statistics_msgs/msg/StatisticDataPoint.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/665/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/StatisticDataPoint.idl
Reading input file: external/ros2_rcl_interfaces/statistics_msgs/msg/StatisticDataType.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/665/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/statistics_msgs/msg/StatisticDataType.idl
INFO: From Action external/ros2_rcl_interfaces/composition_interfaces/rosidl_adapter_map.idls:
Reading input file: external/ros2_rcl_interfaces/composition_interfaces/srv/ListNodes.srv
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/composition_interfaces/srv/ListNodes.idl
Reading input file: external/ros2_rcl_interfaces/composition_interfaces/srv/LoadNode.srv
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/composition_interfaces/srv/LoadNode.idl
Reading input file: external/ros2_rcl_interfaces/composition_interfaces/srv/UnloadNode.srv
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/689/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/composition_interfaces/srv/UnloadNode.idl
INFO: From Action external/ros2_rcl_interfaces/lifecycle_msgs/rosidl_adapter_map.idls:
Reading input file: external/ros2_rcl_interfaces/lifecycle_msgs/msg/State.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/682/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/lifecycle_msgs/msg/State.idl
Reading input file: external/ros2_rcl_interfaces/lifecycle_msgs/msg/Transition.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/682/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/lifecycle_msgs/msg/Transition.idl
Reading input file: external/ros2_rcl_interfaces/lifecycle_msgs/msg/TransitionDescription.msg
Writing output file: /home/liyou/.cache/bazel/_bazel_liyou/e75e8ec54e95eaef5eea312839932295/sandbox/linux-sandbox/682/execroot/com_github_mvukov_rules_ros2/bazel-out/k8-fastbuild/bin/external/ros2_rcl_interfaces/lifecycle_msgs/msg/TransitionDescription.idl
Reading input file: external/ros2_rcl_interfaces/lifecycle_msgs/msg/TransitionEvent.msg
[... 87 lines cut ...]
ImportError: 
Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
If you're working with a numpy git repo, try `git clean -xdf` (removes all
files not under version control).  Otherwise reinstall numpy.

Original error was: cannot import name 'multiarray' from partially initialized module 'numpy.core' (most likely due to a circular import) (/usr/lib/python3/dist-packages/numpy/core/__init__.py)

Exception ignored in: <function Interpreter.__del__ at 0x7f038772c040>
Traceback (most recent call last):
[... cut ...]
ERROR: /home/liyou/work/rules_ros2/examples/chatter/BUILD.bazel:38:12 Middleman _middlemen/examples_Schatter_Schatter-runfiles failed: (Exit 1): rosidl_generator_py_app failed: error executing command bazel-out/k8-opt-exec-2B5CBBC6/bin/external/ros2_rosidl_python/rosidl_generator_py_app ... (remaining 2 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
INFO: Elapsed time: 49.644s, Critical Path: 1.53s
INFO: 108 processes: 81 internal, 27 linux-sandbox.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
```

LiyouZhou, 2022-10-10:
It looks like python3.8 is trying to import an external version of numpy. however when I removed the system number package, the error became `ModuleNotFoundError: No module named 'numpy'`

mvukov, 2022-10-10:
thanks for reporting this. It's time to set up a proper CI for the repo... The bug is fixed in https://github.com/mvukov/rules_ros2/commit/20b8fdf188bec0f8a761a967e47749abedecefd2. Please let me know if this solves issues for you.

mvukov, 2022-10-12:
And another fix is in [921a483](https://github.com/mvukov/rules_ros2/commit/921a4836a39304b85c22ce7fcbb3b03a341e65de).

closed 2022-10-15 by mvukov