mvukov/rules_ros2 #64: Using `--nolegacy_external_runfiles` will break loading `librmw_cyclonedds.so`

lalten, 2023-02-28:
If I add `--nolegacy_external_runfiles` to my .bazelrc as recommended in https://docs.aspect.build/guides/bazelrc/#performance-options (see below)
```sh
# Do not build runfiles symlink forests for external repositories under
# `.runfiles/wsname/external/repo` (in addition to `.runfiles/repo`). This reduces runfiles &
# sandbox creation times & prevents accidentally depending on this feature which may flip to off by
# default in the future. Note, some rules may fail under this flag, please file issues with the rule
# author.
# Docs: https://bazel.build/reference/command-line-reference#flag--legacy_external_runfiles
build --nolegacy_external_runfiles
run --nolegacy_external_runfiles
test --nolegacy_external_runfiles
```

Then code depending on rclcpp will fail with
```
C++ exception with description "failed to initialize rcl init options: failed to load shared library 'external/ros2_rmw_cyclonedds/librmw_cyclonedds.so' due to dlopen error: external/ros2_rmw_cyclonedds/librmw_cyclonedds.so: cannot open shared object file: No such file or directory, at external/ros2_rcutils/src/shared_library.c:99, at external/ros2_rmw_implementation/rmw_implementation/src/functions.cpp:90, at external/ros2_rcl/rcl/src/rcl/init_options.c:75" thrown in the test body.
```

This seems very related to https://github.com/mvukov/rules_ros2/issues/57

ahans, 2023-02-28:
I put up a PR to fix this, see links above. While this makes it work with `--nolegacy_external_runfiles`, the overall setup is still somewhat fragile and very much relies on the way Bazel lays out data in the rundir. As commented over in #57, I think it'd be better to move closer to the standard ROS 2 setup. But that's certainly something for another time ... ;-)

mvukov, 2023-02-28:
Well, #57 is still open for a discussion. :)

closed 2023-02-28 by mvukov

mvukov, 2023-03-01:
BTW, not all rule sets can work with `--nolegacy_external_runfiles` yet. But you'll see errors in such cases.