mvukov/rules_ros2 #95: Runtime error when using rosbag2_cpp

kgreenek, 2023-04-13:
Hi! I am loving this project so far, but I have run into a bit of a snag.

I created a small C++ binary with the goal of reading from a ros2 bag file and writing to a CSV file. However, when I try to run my binary I get the following error:

```
[ERROR] [1681401142.875500463] [rosbag2_storage]: Unable to create class load instance: Environment variable 'AMENT_PREFIX_PATH' is not set or empty
```

Here is my BUILD.bazel file:

```bzl
load("@com_github_mvukov_rules_ros2//ros2:cc_defs.bzl", "ros2_cpp_binary")

ros2_cpp_binary(
    name = "ros2_nav_sat_fix_to_csv",
    srcs = ["main.cpp"],
    deps = [
        "@ros2_rclcpp//:rclcpp",
        "@ros2_rosbag2//:rosbag2_cpp",
        "@ros2_common_interfaces//:cpp_sensor_msgs",
    ],
)
```

And here is my main.cpp:

```c++
#include <fstream>
#include <iostream>

#include "rosbag2_cpp/reader.hpp"
#include "sensor_msgs/msg/nav_sat_fix.hpp"

namespace {

void WriteCsvHeader(std::ofstream& csv_fstream) {
  csv_fstream << "lat,lon" << std::endl;
}

void WriteCsvPosition(std::ofstream& csv_fstream,
                      const sensor_msgs::msg::NavSatFix& nav_sat_fix_msg) {
  csv_fstream << nav_sat_fix_msg.latitude << nav_sat_fix_msg.longitude << std::endl;
}

}  // namespace

int main(int argc, char** argv) {
  if (argc != 3) {
    std::cerr << "Usage: " << argv[0] << " [input bag path] [output csv path]" << std::endl;
    return 1;
  }
  const std::string in_bag_path{argv[1]};
  const std::string out_csv_path{argv[2]};
  rosbag2_cpp::Reader in_bag_reader;
  in_bag_reader.open(in_bag_path);
  std::ofstream csv_fstream{out_csv_path, std::ofstream::out | std::ofstream::trunc};
  if (!csv_fstream) {
    std::cerr << "Could not open output CSV" << std::endl;
    return 1;
  }

  WriteCsvHeader(csv_fstream);
  while (in_bag_reader.has_next()) {
    const auto nav_sat_fix_msg = in_bag_reader.read_next<sensor_msgs::msg::NavSatFix>();
    WriteCsvPosition(csv_fstream, nav_sat_fix_msg);
  }

  return 0;
}
```

ahans, 2023-04-13:
You need to add `set_up_ament = False` to the `ros2_cpp_binary` call. Then a shell wrapper will be created that sets the ament env variable.

closed 2023-04-13 by kgreenek

mvukov, 2023-04-17:
> Question for you: do you think it makes sense for this to be the default option?

Ament setup is only needed when you e.g. use plugins. Otherwise, it's not necessary. Therefore, I am hesitant to set it to true per default.

cssharsha, 2024-12-24:
@kgreenek, @mvukov Do you happen to have a complete example somewhere? I have a similar setup where Im rading from a bag and creating a rerun_bridge to visualize it. The `set_up_ament` does fix the `AMENT_PREFIX_PATH` not set error, but ended up having a different error where it complains that the *_impl is not present for the target I build using `ros2_cpp_binary`

cssharsha, 2024-12-25:
Here's an example I'm trying out:

A simple main reading a bag and listing the topics in the bag:
```
#include <iostream>
#include <rosbag2_cpp/reader.hpp>
#include <rosbag2_storage/storage_options.hpp>

int main(int argc, char** argv) {
    if (argc != 2) {
        std::cerr << "Usage: " << argv[0] << " <path_to_rosbag>" << std::endl;
        return 1;
    }

    try {
        rosbag2_storage::StorageOptions storage_options;
        storage_options.uri = argv[1];
        storage_options.storage_id = "sqlite3";

        rosbag2_cpp::Reader reader;
        reader.open(storage_options);

        std::cout << "\nTopics in bag file:" << std::endl;
        std::cout << "==================" << std::endl;

        for (const auto& topic : reader.get_all_topics_and_types()) {
            std::cout << "Topic: " << topic.name << std::endl;
            std::cout << "Type: " << topic.type << std::endl;
            std::cout << "-------------------" << std::endl;
        }

    } catch (const std::exception& e) {
        std::cerr << "Error: " << e.what() << std::endl;
        return 1;
    }

    return 0;
}
```

The corresponding build file:
```
load("@com_github_mvukov_rules_ros2//ros2:cc_defs.bzl", "ros2_cpp_binary")

ros2_cpp_binary(
    name = "list_topics",
    srcs = ["examples/list_topics.cpp"],
    deps = [
        "@ros2_rosbag2//:rosbag2_cpp",
        "@ros2_rosbag2//:rosbag2_storage",
    ],
    set_up_ament = True,
)
```

It generates two targets:
```
➜ /workspace (viz) ✗ bazel build //viz:list_topics
INFO: Analyzed target //viz:list_topics (111 packages loaded, 13615 targets configured).
INFO: From Compiling sqlite3.c:
external/sqlite3/sqlite3.c:110248:38: warning: implicit conversion from 'long long' to 'double' changes value from 9223372036854775806 to 9223372036854775808 [-Wimplicit-const-int-float-conversion]
  if( n==0 && r>=0 && r<LARGEST_INT64-1 ){
                       ~~~~~~~~~~~~~~^~
external/sqlite3/sqlite3.c:110250:46: warning: implicit conversion from 'long long' to 'double' changes value from 9223372036854775806 to 9223372036854775808 [-Wimplicit-const-int-float-conversion]
  }else if( n==0 && r<0 && (-r)<LARGEST_INT64-1 ){
                               ~~~~~~~~~~~~~~^~
2 warnings generated.
INFO: Found 1 target...
Target //viz:list_topics up-to-date:
  bazel-bin/viz/list_topics_launch
  bazel-bin/viz/list_topics
INFO: Elapsed time: 37.202s, Critical Path: 16.67s
INFO: 633 processes: 191 internal, 7 local, 435 processwrapper-sandbox.
INFO: Build completed successfully, 633 total actions
```

Getting the `*_impl` not found error when trying to run either of the binaries:
```
➜ /workspace (viz) ✗ ./bazel-bin/viz/list_topics_launch 
./bazel-bin/viz/list_topics_launch: line 20: /workspace/viz/list_topics_impl: No such file or directory
➜ /workspace (viz) ✗ ./bazel-bin/viz/list_topics       
./bazel-bin/viz/list_topics: line 20: /workspace/viz/list_topics_impl: No such file or directory
```

When I try to run the `_impl` directly I get the `AMENT_PREFIX_PATH` not loaded error:
```
➜ /workspace (viz) ✗ ./ba./bazel-bin/viz/list_topics_impl /mnt/remote-storage/all_topics_2024_12_05_19_23_52/all_topics_2024_12_05_19_23_52_0.db3 
[ERROR] [1735149281.233687720] [rosbag2_storage]: Unable to create class load instance: Environment variable 'AMENT_PREFIX_PATH' is not set or empty
Error: std::exception
```

My `.bazelrc` and `WORKSPACE` can be found [here](https://github.com/cssharsha/shadesmar/blob/viz/.bazelrc) and [here](https://github.com/cssharsha/shadesmar/blob/viz/WORKSPACE).
_PS: Still in initial phase so might be extremely dirty to look at._

mvukov, 2024-12-25:
Long story in short, `./bazel-bin/viz/list_topics_launch` isn't going to work (you need to `cd` within the relevant .runfiles folder). A simple way to make this work is `bazel run //viz:list_topics`. If you need to deploy your app to run without Bazel that's a different story -- it's def possible to make this work.

cssharsha, 2024-12-25:
Yeah for now, Im just using `bazel run`. Wanted these binary files to be generated and be run in a non bazel env. Looks like that might be quite a hassle with ros2 plugin loads. Ill just let them be in bazel for now. Will try to figure that out sometime way down the line.

Thanks for the replies and the help!