mvukov/rules_ros2 #97: Feature request: Support tf2

kgreenek, 2023-04-14:
Title is self-explanatory. I believe this is where the code lies (along with some other geometry-related ros2 libraries):
https://github.com/ros2/geometry2

Is this already on your radar? If not, is this something you would be interested in supporting in this repo, or would it be more appropriate as a separate repo?

ahans, 2023-04-14:
It is largely already there. This is the BUILD file: https://github.com/mvukov/rules_ros2/blob/main/repositories/geometry2.BUILD.bazel

It's imported here: https://github.com/mvukov/rules_ros2/blob/main/repositories/ros2_repositories_impl.bzl#L53

Do a `bazel query @ros2_geometey2//...` to get a list of the available targets.

kgreenek, 2023-04-14:
Oh I just didn't see it it. Yay! Thank you!

closed 2023-04-14 by kgreenek