mvukov/rules_ros2 #115: How to launch a robot_state_publisher node

armuren, 2023-05-25:
Many thanks for this excellent work.  It's a great contribution.

I was able to build the `robot_state_publisher` package as in the following (please note it's a little bit messy at the moment)

```
ros2_cpp_library(
    name = "ros2_robot_localization",
    srcs = glob([
        "src/*.cpp",
    ]),
    hdrs = glob(["include/**/*.hpp",]),
    includes = ["include"],
    visibility = ["//visibility:public"],
    deps = [
        ":cpp_localization_msgs",

        "@ros2_rclcpp//:rclcpp",
        "@ros2_rcutils//:rcutils",

        "@ros2_common_interfaces//:cpp_nav_msgs",
        "@ros2_common_interfaces//:cpp_sensor_msgs",
        "@ros2_common_interfaces//:cpp_std_msgs",
        "@ros2_common_interfaces//:cpp_std_srvs",
        "@ros2_common_interfaces//:cpp_diagnostic_msgs",

        "@ros2_geographic_msgs//:cpp_geographic_msgs",
        "@ros2_common_interfaces//:cpp_geometry_msgs",

        "@ros2_geometry2//:tf2",
        "@ros2_geometry2//:tf2_ros",
        "@ros2_geometry2//:tf2_eigen",
        "@ros2_geometry2//:tf2_geometry_msgs",
        "@ros2_geometry2//:cpp_tf2_msgs",

        "@yaml_cpp",
        "@eigen",
        "@boost",
        "@GeographicLib",
        "@ros2_diagnostics",
        "@ros2_angles",
    ],
)
```

I then attempted to launch a node from that package using the standard launch file technique

```
# hard coding the AMENT_PREFIX_PATH for now
os.environ["AMENT_PREFIX_PATH"] = "/home/ubuntu/.cache/bazel/_bazel_ubuntu/e9106c449e78ffee07178e6c7277bb20/external/"
# standard launch file similar to chatter ...
robot_state_publisher_node = launch_ros.actions.Node(
          package='robot_state_publisher',
          executable='robot_state_publisher',
          parameters=[{'robot_description': Command(['xacro ', LaunchConfiguration('model')])}]
      )
```

and a BUILD file as:

```
ros2_launch(
    name = "nav2_bot",
    launch_file = "nav2_bot.py",
    nodes = [], # I had to make a change to ros2_launch.bzl to allow empty nodes
    data = [
        "my_world.sdf",
        "nav2_bot_description.urdf",
        "urdf_config.rviz",
    ],
    deps = [
        #"@ros2_robot_state_publisher",
    ]
)
```

However `bazel run //nav2_bot` still says the robot_state_publisher package is not found.  I called `bazel build @ros2_robot_state_publisher` it to verify I have that setup correctly and it worked.  I can also find it in the bazel output directory as well as it's `.so` files.

My guess I am missing something simple?

armuren, 2023-05-25:
I see in the `chatter.py` that the executable should refer to the directory path of the bin.

mvukov, 2023-05-27:
`package` in launch_ros.actions.Node doesn't have a meaning any more when you use Bazel. You can just forget about it. In `executable`, on the other hand, you should put an exec path to your executable. The easiest is to create a template like in https://github.com/mvukov/rules_ros2/blob/main/examples/zero_copy/BUILD.bazel#L48-L62 and then don't really think about where the executable lives. Also, you don't need to handle AMENT_PREFIX_PATH directly, it will be set for you automatically. Lemme know if you need more help. In case still having problems, it would be helpful to set up a simple repo with your code such that I can get more insights and provide possibly better feedback.

armuren, 2023-05-27:
Thanks @mvukov .  What if I am trying to run a third party package, such as `robot_state_publisher` which a registered component[ similar to ](https://github.com/ros/robot_state_publisher/blob/humble/CMakeLists.txt#L48)

```
rclcpp_components_register_node(${PROJECT_NAME}_node
  PLUGIN "robot_state_publisher::RobotStatePublisher"
  EXECUTABLE robot_state_publisher)
```

I already have bazel build of robot_state_publishe up in [Github](https://github.com/xiaocars/xiaocars/blob/issue-27/create-odom-and-use-ros2/third_party/rules_ros2/repositories/ros2_robot_state_publisher.BUILD.bazel), however I have to warn you, it's a bit messy still, and I've have to hard code a few things because I am developing on x86_64 but testing on an Nvidia Jetson arm64.  Everything works fine in terms of building and running your chatter example on both platforms actually, but with some messy code that requires cleanup.

mvukov, 2023-05-28:
There is no support for components, yet. A node generated by rclcpp_components_register_node won't work AFAIK as it tries to load the component plugin.

armuren, 2023-05-28:
Thanks @mvukov that's what I sort of expected.  So when creating large systems using the current approach, the best practice would be to write some kind of executable that registers the components? Similar to the [manual composition ](https://github.com/ros2/demos/blob/rolling/composition/src/manual_composition.cpp) example from ros2 demos repo? Or is there a better approach?

I believe the robot_state_publisher node can be [subclassed](http://wiki.ros.org/robot_state_publisher/Tutorials/Using%20the%20robot%20state%20publisher%20on%20your%20own%20robot#Running_as_a_library) as the example shows in the documentation.  However I have not tried this myself previously.

mvukov, 2023-05-28:
IMO, manual composition in a single executable is fine; but it depends a bit what you want. You can read e.g. https://arxiv.org/abs/2305.09933 on some best practices.

You can include https://github.com/ros/robot_state_publisher/blob/humble/include/robot_state_publisher/robot_state_publisher.hpp in your executable, instantiate a node and add it to an executor.

armuren, 2023-05-28:
Thanks @mvukov .  I appreciate your time on this work and your effort in answering my questions.  Ultimately while I would prefer this approach to the traditional build process, it requires an entire team of dedicated build engineers, which is something I don't have access to at the moment.

closed 2023-05-28 by armuren

mvukov, 2023-05-29:
Feel free to share your experience here such that we can improve DevEx.