mvukov/rules_ros2 #127: Runtime exception when using node->create_generic_subscription

kgreenek, 2023-06-23:
I am trying to use the `rclcpp::Node::create_generic_subscription` method to create a subscription that does not need to be aware of the type of the message at compile time.

My use-case is a simple data recorder node that takes a list of topics as input via config. It subscribes to all of those topics and dumps them into bag files in a pre-determined directory hierarchy.

Here is a stripped down example repo that reproduces the exception:
https://github.com/kgreenek/rules_ros2_typesupport_bug/blob/main/example/main.cpp#L4

I am getting the following runtime error:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Typesupport library for sensor_msgs does not exist in 'example/example_launch_ament_setup'.
/home/kgk/.cache/bazel/_bazel_kgk/25cdd435d42a5ead8f097b62940dbe88/execroot/kgkslam/bazel-out/k8-fastbuild/bin/example/example: line 21: 1052752 Aborted                 (core dumped) AMENT_PREFIX_PATH="${ament_prefix_path}" example/example_impl "$@"
```

In that repo, you can run `bazel run example` to see the crash.

Honestly I think it's pretty likely I'm doing something wrong here, but it also seems like it might be an unsupported feature. I'd really appreciate your help!

ahans, 2023-06-23:
In your `ros2_cpp_binary`, you need to give the IDL deps explicitly via `idl_deps`, otherwise the type support libraries won't be put into the ament setup:
```Python
ros2_cpp_binary(
    name = "example",
    srcs = ["main.cpp"],
    idl_deps = [
        "@ros2_common_interfaces//:sensor_msgs",
        "@ros2_common_interfaces//:std_msgs",
    ],
    set_up_ament = True,
    deps = [
        "@ros2_rclcpp//:rclcpp",
    ],
)
```
You only need the `cpp_*` ones in your `deps` if you want to directly from your C++ code use the type support code. Then it will be linked into your binary and you don't need the `idl_deps` for this. But for runtime dependencies that are unknown at compile time (at least from a C++ perspective, from Bazel perspective they must be known), you need to give the `idl_deps`, so the type support `.so` files are put into the ament tree.

kgreenek, 2023-06-23:
Oh! Thank you! That was the problem! I added the appropriate idl_deps and verified that now I can use the generic subscription :)

closed 2023-06-23 by kgreenek

mvukov, 2023-06-24:
Great that this is already resolved. There was/is a test with this setup, see https://github.com/mvukov/rules_ros2/blob/main/ros2/test/BUILD.bazel#L61.