mvukov/rules_ros2 #167: Unable to build using Zig hermetic toolchain (linker cannot find library -latomic)

tervay-bdai, 2023-08-21:
Hi, thanks for your work on this, it is overall a great library.

We are using [Uber's hermetic CC toolchain](https://github.com/uber/hermetic_cc_toolchain) to compile and are unable to get simple python scripts depending on rclpy to build. A minimal reproduction can be found here: https://github.com/tervay-bdai/zig-linking-repro

I am truthfully not sure if this problem lies within the toolchain or within rules_ros2. It seems that when building with this toolchain, it's not able to find the atomic library, even though mentions of it seem to be found in the generated bazel directories.

The line from rules_ros2 causing the problem is here: https://github.com/mvukov/rules_ros2/blob/a3e124988e8215f505bee467c73074e6f7c72dee/repositories/rclpy.BUILD.bazel#L21

I've tried patching the flag out, but it results in an illegal operation runtime error, so I assume the flag is actually required and patching it out doesn't actually solve anything.

Guidance would be appreciated here, if any; I can also file an issue on the Zig toolchain repo but again, I'm not sure where the problem actually is.

mvukov, 2023-08-22:
Hi, on Ubuntu the library is provided via apt in `libatomic1` package. I guess you also need to have this library in the rootfs the compiler works on.

mvukov, 2023-09-01:
Any progress here?

mvukov, 2023-09-13:
Can it be that with zig/cc you don't need to link `-latomic`, like is the case with android/qnx?

tervay-bdai, 2023-09-13:
Unfortunately no, no progress. Had to revert to clang toolchains in the meantime to unblock other things.

mvukov, 2023-09-23:
OK, let's close this for now then.

closed 2023-09-23 by mvukov