mvukov/rules_ros2 #184: Foxglove bridge Build error

sibi-venti, 2023-10-01:
```
 Linking third_party/foxglove_bridge/foxglove_bridge_node failed: (Exit 1): gcc failed: error executing command (from target //third_party/foxglove_bridge:foxglove_bridge_node) /usr/bin/gcc @bazel-out/k8-fastbuild/bin/third_party/foxglove_bridge/foxglove_bridge_node-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_session_set_blocking'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_session_handshake'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_hostkey_hash'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_hostkey_hash'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_knownhost_free'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_agent_disconnect'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_agent_free'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_session_free'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_channel_free'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_session_disconnect_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_channel_free'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_channel_wait_closed'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_channel_wait_eof'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_channel_send_eof'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_scp_recv2'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_scp_send64'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_symlink_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_readdir_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_mkdir_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_open_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_seek64'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_close_handle'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_close_handle'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_shutdown'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_open_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_close_handle'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_stat_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_init'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_userauth_keyboard_interactive_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_agent_list_identities'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_agent_connect'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_agent_get_identity'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_userauth_password_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_userauth_publickey_fromfile_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_userauth_list'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_symlink_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_stat_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_last_error'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_unlink_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_last_error'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_rmdir_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_last_error'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_rename_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_last_error'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_mkdir_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_statvfs'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_stat_ex'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function ssh_statemach_act: error: undefined reference to 'libssh2_sftp_stat_ex'
[... cut ...]
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function Curl_ssh_version: error: undefined reference to 'libssh2_version'
bazel-out/k8-fastbuild/bin/external/curl/curl/lib/libcurl.a(libssh2.c.o):libssh2.c:function Curl_ssh_cleanup: error: undefined reference to 'libssh2_exit'
collect2: error: ld returned 1 exit status
INFO: Elapsed time: 283.168s, Critical Path: 260.66s
INFO: 2710 processes: 1393 internal, 1317 linux-sandbox.
FAILED: Build did NOT complete successfully

```

how do i solve this?

mvukov, 2023-10-01:
Hi,

It looks like SSH2 is enabled by default https://github.com/curl/curl/blob/master/CMakeLists.txt#L876C8-L876C24. Please try to add `"CURL_USE_LIBSSH2": "OFF"` to https://github.com/mvukov/rules_ros2/blob/main/repositories/curl.BUILD.bazel#L18. Don't know why you get those linking errors, see for instance `bazel-bin/external/curl/curl_foreign_cc/CMake.log` to learn more.

sibi-venti, 2023-10-03:
```
    cache_entries = {
        "CMAKE_POSITION_INDEPENDENT_CODE": "ON",  # Must be set!
        "BUILD_SHARED_LIBS": "OFF",
        # curl specific options.
        "BUILD_CURL_EXE": "OFF",
        "CMAKE_PREFIX_PATH": "$EXT_BUILD_DEPS/openssl",
        "CURL_DISABLE_LDAP": "ON",
        "CURL_ENABLE_SSL": "ON",
        "CURL_USE_LIBPSL": "OFF",
        "CURL_USE_LIBSSH2": "OFF",
        # TODO(mvukov) Fine-tune other options.
    },
```
    it worked https://github.com/mvukov/rules_ros2/blob/main/repositories/curl.BUILD.bazel#L18

closed 2023-10-03 by sibi-venti