mvukov/rules_ros2 #191: undefined reference to 'idn2_check_version', 'idn2_lookup_ul', 'idn2_XXXXXX'...

cdelguercio, 2023-10-13:
Ubuntu 22.04
gcc 11.4.0
I have libidn2-0 and libidn2-0-dev installed

My colleague has the same setup and it builds fine on their machine. My system was installed fresh about a week ago so I doubt anything too weird is going on in my environment.

```
ERROR: ...external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/BUILD.bazel:11:16: Linking external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/foxglove_bridge_node failed: (Exit 1): gcc failed: error executing command (from target @com_github_mvukov_rules_ros2//third_party/foxglove_bridge:foxglove_bridge_node) /usr/bin/gcc @bazel-out/k8-opt/bin/external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/foxglove_bridge_node-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
...external/curl/lib/idn.c:123: error: undefined reference to 'idn2_check_version'
...external/curl/lib/idn.c:132: error: undefined reference to 'idn2_lookup_ul'
...external/curl/lib/idn.c:153: error: undefined reference to 'idn2_free'
...external/curl/lib/idn.c:136: error: undefined reference to 'idn2_lookup_ul'
...external/curl/lib/idn.c:167: error: undefined reference to 'idn2_free'
...external/curl/lib/idn.c:123: error: undefined reference to 'idn2_check_version'
...external/curl/lib/idn.c:132: error: undefined reference to 'idn2_lookup_ul'
...external/curl/lib/idn.c:136: error: undefined reference to 'idn2_lookup_ul'
...external/curl/lib/idn.c:189: error: undefined reference to 'idn2_free'
collect2: error: ld returned 1 exit status
```

WORKSPACE.bazel
```
http_archive(
    name = "com_github_mvukov_rules_ros2",
    sha256 = "2436c86f7830eb6965829db020eb33c41e369401b567aca1a8e0cd1eaf7cb772",
    strip_prefix = "rules_ros2-32d18c6b45897e0e818d1386eb578b21b8a29d7e",
    urls = ["https://github.com/mvukov/rules_ros2/archive/32d18c6b45897e0e818d1386eb578b21b8a29d7e.tar.gz"],
)

load("@com_github_mvukov_rules_ros2//repositories:repositories.bzl", "ros2_repositories")

ros2_repositories()

load("@com_github_mvukov_rules_ros2//repositories:deps.bzl", "ros2_deps")

ros2_deps()

load(
    "@com_github_mvukov_rules_ros2//repositories:pip_annotations.bzl",
    ROS2_PIP_ANNOTATIONS = "PIP_ANNOTATIONS",
)

pip_parse(
    name = "rules_ros2_pip_deps",
    annotations = ROS2_PIP_ANNOTATIONS,
    python_interpreter_target = python_interpreter_target,
    requirements_lock = "@com_github_mvukov_rules_ros2//:requirements_lock.txt",
)

load(
    "@rules_ros2_pip_deps//:requirements.bzl",
    install_rules_ros2_pip_deps = "install_deps",
)

install_rules_ros2_pip_deps()
```

mvukov, 2023-10-14:
Please check a fix in https://github.com/mvukov/rules_ros2/pull/193.

cdelguercio, 2023-10-16:
Seems like some progress has been made, but now I'm getting the following:
```
ERROR: ...external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/BUILD.bazel:11:16: Linking external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/foxglove_bridge_node failed: (Exit 1): gcc failed: error executing command (from target @com_github_mvukov_rules_ros2//third_party/foxglove_bridge:foxglove_bridge_node) /usr/bin/gcc @bazel-out/k8-opt/bin/external/com_github_mvukov_rules_ros2/third_party/foxglove_bridge/foxglove_bridge_node-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
...external/curl/lib/vtls/openssl.c:1627: error: undefined reference to 'EVP_PKEY_get_id'
...external/curl/lib/vtls/openssl.c:510: error: undefined reference to 'EVP_PKEY_get_id'
...external/curl/lib/vtls/openssl.c:531: error: undefined reference to 'EVP_PKEY_get_bn_param'
...external/curl/lib/vtls/openssl.c:532: error: undefined reference to 'EVP_PKEY_get_bn_param'
...external/curl/lib/vtls/openssl.c:567: error: undefined reference to 'EVP_PKEY_get_bn_param'
...external/curl/lib/vtls/openssl.c:568: error: undefined reference to 'EVP_PKEY_get_bn_param'
...external/curl/lib/vtls/openssl.c:4078: error: undefined reference to 'SSL_get1_peer_certificate'
...external/curl/lib/vtls/openssl.c:2414: error: undefined reference to 'SSL_get1_peer_certificate'
...external/curl/lib/vtls/openssl.c:3189: error: undefined reference to 'X509_STORE_load_file'
...external/curl/lib/vtls/openssl.c:3198: error: undefined reference to 'X509_STORE_load_path'
...external/curl/lib/vtls/openssl.c:3198: error: undefined reference to 'X509_STORE_load_path'
```

WORKSPACE.bazel change:
```
# Latest commit as of: Oct 15, 2023
http_archive(
    name = "com_github_mvukov_rules_ros2",
    sha256 = "246c069c6acaa9ed0b1860e6496d85253c76d295a50904b4cf1b7b647d9af581",
    strip_prefix = "rules_ros2-063336d70f770f0ca02f246912424f5e97bf1393",
    urls = ["https://github.com/mvukov/rules_ros2/archive/063336d70f770f0ca02f246912424f5e97bf1393.tar.gz"],
)
```

mvukov, 2023-10-16:
Interesting. Please provide a minimal failing build with a Dockerfile such that I can reproduce the issue. Please also attach here `bazel-bin/external/curl/curl_foreign_cc/CMake.log` from your failing build.

kgreenek, 2023-10-18:
I am able to reproduce the exact EVP key linker errors that Chris is seeing by building the foxglove_bridge example in this repo. I.E.

```
cd examples
bazel build foxglove_bridge/...
```

I wonder if it has something to do with openssl being updated upstream.

One idea is to try switching to the hedronvision fork of boringssl, which has bazel BUILD files already. That is the implementation that boost uses as well, so it would simultaneously address this issue: https://github.com/mvukov/rules_ros2/issues/197

kgreenek, 2023-10-18:
I went ahead and tried that, and it appears to work! I'll create a PR shortly to look at.

closed 2023-10-22 by mvukov