mvukov/rules_ros2 #283: `ros2_interface_library` generate wrong message include path for action and srv

zhuhaow, 2024-03-02:
I created a minimal repro at https://github.com/zhuhaow/ros2-bazel-message-error-demo/

When running `bazel build --keep_going //...`, we will get error

```
ERROR: /home/zhuhaow/demo/example/BUILD.bazel:57:17: Compiling example/example_action.cpp failed: (Exit 1): gcc failed: error executing CppCompile command (from target //example:example_action_lib) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 195 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/example/example_action/action/example.hpp:7,
                 from example/example_action.cpp:1:
bazel-out/k8-fastbuild/bin/example/example_action/action/detail/example__struct.hpp:20:10: fatal error: example_action/msg/detail/example__struct.hpp: No such file or directory
   20 | #include "example_action/msg/detail/example__struct.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ERROR: /home/zhuhaow/demo/example/BUILD.bazel:46:17: Compiling example/example_srv.cpp failed: (Exit 1): gcc failed: error executing CppCompile command (from target //example:example_srv_lib) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 181 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/example/example_srv/srv/example.hpp:7,
                 from example/example_srv.cpp:1:
bazel-out/k8-fastbuild/bin/example/example_srv/srv/detail/example__struct.hpp:20:10: fatal error: example_srv/msg/detail/example__struct.hpp: No such file or directory
   20 | #include "example_srv/msg/detail/example__struct.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

The problem seems like the message header should be `example_msg/msg/detail/example__struct.hpp` but somehow it is generated as the target depends on it `{TARGET_NAME}/msg/detail/example__struct.hpp`

mvukov, 2024-03-05:
So, Bazel package name != ros_package_name and they are unrelated. But, ros_package_name == target name. In your case, you defined 3 separate targets (please share why) example_msg, example_srv and example_action. Those will defined 3 separate packages in ROS terms. Therefore, you need to the following changes:

```
diff --git a/example/Example.action b/example/Example.action
index c0b78f8..be916b6 100644
--- a/example/Example.action
+++ b/example/Example.action
@@ -1,5 +1,5 @@
-Example message1
+example_msg/Example message1
 ---
-Example message2
+example_msg/Example message2
 ---
-Example message3
+example_msg/Example message3
diff --git a/example/Example.srv b/example/Example.srv
index d0bba73..ab7d8a2 100644
--- a/example/Example.srv
+++ b/example/Example.srv
@@ -1,3 +1,3 @@
-Example ping
+example_msg/Example ping
 ---
-Example pong
+example_msg/Example pong
```
With those changes, you can build the other cc-targets. Sounds good?

zhuhaow, 2024-03-06:
That makes total sense! Thanks so much

closed 2024-03-06 by zhuhaow