mvukov/rules_ros2 #405: What is the correct target to specify dependancy on for rosbag handler cpp tools

cssharsha, 2024-12-12:
I have a c++ tool that reads a particular bag file and parses the corresponding sensor messages. Im adding a cc_test that mocks a bagfile where it creates one. Im using the `rosbag2_cpp/writers/sequential_writer` to write to a bag file.

The write function is leading to a linker error. Below is the code snippet:

```
rosbag2_cpp::Writer writer;
writer.open(test_bag_path_);
auto odom_msg = std::make_shared<nav_msgs::msg::Odometry>();
odom_msg->header.stamp = time;
odom_msg->header.frame_id = "odom";
odom_msg->child_frame_id = "base_link";
odom_msg->pose.pose.position.x = i * 0.1;
odom_msg->pose.pose.orientation.w = 1.0;
writer.write(odom_msg, config_.odom_topic, time); // linker error here
```

Below is the actual error I get:
```
ERROR: /workspaces/shadesmar/ros/BUILD:49:8: Linking ros/rosbag_reader_test failed: (Exit 1): clang failed: error executing CppLink command (from target //ros:rosbag_reader_test) /usr/lib/llvm-14/bin/clang @bazel-out/k8-fastbuild/bin/ros/rosbag_reader_test-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
ld.lld: error: undefined symbol: char const* rosidl_generator_traits::name<std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(void rosbag2_cpp::Writer::write<std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > > >(std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::Time const&))

ld.lld: error: undefined symbol: char const* rosidl_generator_traits::name<std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(void rosbag2_cpp::Writer::write<std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > > >(std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::Time const&))

ld.lld: error: undefined symbol: char const* rosidl_generator_traits::name<std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(void rosbag2_cpp::Writer::write<std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > >(std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::Time const&))

ld.lld: error: undefined symbol: rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(rclcpp::Serialization<std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > > >::Serialization())

ld.lld: error: undefined symbol: rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(rclcpp::Serialization<std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > > >::Serialization())

ld.lld: error: undefined symbol: rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > >()
>>> referenced by rosbag_reader_test.cpp
>>>               bazel-out/k8-fastbuild/bin/ros/_objs/rosbag_reader_test/rosbag_reader_test.pic.o:(rclcpp::Serialization<std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > >::Serialization())
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Any indication which target to add as a `dep` would be very helpful. I guess you might need more context to resolve this error. Please let me know.

cssharsha, 2024-12-12:
BUILD file for reference:
```
load("@rules_cc//cc:defs.bzl", "cc_test")
load("@com_github_mvukov_rules_ros2//ros2:cc_defs.bzl", "ros2_cpp_library")

ros2_cpp_library(
    name = "ros_conversions",
    srcs = ["src/conversions.cpp"],
    hdrs = ["include/ros/conversions.hpp"],
    includes = ["include"],
    deps = [
        "//core:core",
        "@ros2_rclcpp//:rclcpp",                # ROS2 C++ client library
        "@ros2_common_interfaces//:cpp_geometry_msgs",         # Geometry messages
        "@ros2_common_interfaces//:cpp_sensor_msgs",           # Sensor messages
        "@ros2_common_interfaces//:cpp_nav_msgs",             # Navigation messages
        "@ros2_common_interfaces//:cpp_std_msgs",             # Standard messages
        "@ros2_message_filters//:message_filters",       # Message filtering
        "@opencv//:core",
        "@opencv//:imgproc",
    ],
    visibility = ["//visibility:public"],
)

ros2_cpp_library(
    name = "rosbag_reader",
    srcs = ["src/rosbag_reader.cpp"],
    hdrs = ["include/ros/rosbag_reader.hpp"],
    includes = ["include"],
    deps = [
        "//core:core",
        ":ros_conversions",
        "@ros2_rosbag2//:rosbag2_cpp",          # ROS2 bag reading
        "@ros2_message_filters//:message_filters",
    ],
    visibility = ["//visibility:public"],
)

# Test targets
cc_test(
    name = "conversions_test",
    srcs = ["test/conversions_test.cpp"],
    deps = [
        ":ros_conversions",
        "@com_google_googletest//:gtest_main",
        "@ros2_rosbag2//:rosbag2_storage",
    ],
)

cc_test(
    name = "rosbag_reader_test",
    srcs = ["test/rosbag_reader_test.cpp"],
    deps = [
        ":rosbag_reader",
        "@com_google_googletest//:gtest_main",
    ],
)

test_suite(
    name = "ros_tests",
    tests = [
        ":conversions_test",
        ":rosbag_reader_test",
    ],
)
```

cssharsha, 2024-12-13:
I ended up using `ros2_cpp_test` for those tests that required using idl and other plugin stuff. Had not used earlier due to `whl_filegroup` target not being present in the version of  `rules_python` that I was using.

For future use these are the working versions I tested with:
```
rules_cc: 0.0.9
rules_python: 0.36.0
commit used for this: 19d0a447d5d1226897ae032ca6f31548c0b02075
```

closed 2024-12-13 by cssharsha

mvukov, 2024-12-20:
rosbag2 heavily depends on plugins, thus, you need to use ros2_cpp_test (as you figured out in the meantime).