mvukov/rules_ros2 #458: Allow customization of Python rules

ahans, 2025-03-20:
I have a use case where I'd like to use the `py_binary` from [rules_py](https://github.com/aspect-build/rules_py) as the backend of `ros2_py_binary` instead of the default one from `rules_python`. My current solution is a local patch. Would there be any issues with allowing customization from the user? We could probably simply expose the `target` parameter and default it to `None`. If it's set, we'll use it. Otherwise, we use the variant from `rules_python` just like today. What do you think? Am I missing anything? If not, I would put up a PR with this change.

mvukov, 2025-03-23:
Fine by me, please create a PR.

mvukov, 2025-03-23:
Just curious, why do you patch this repo at the moment? You could juse pass target argument to https://github.com/mvukov/rules_ros2/blob/main/ros2/py_defs.bzl#L8C19-L8C25.

PR #460 (2025-06-07): Allow customization of underlying rule in `ros2_py_{binary,test}`
This is what I had in mind. I can add an example where we use `rules_py`, but wasn't sure whether pulling in `rules_py` just for this is justified.

Fixes #458
closed 2025-06-07 by mvukov