mvukov/rules_ros2 #499: Support converting .idl to cc_library

Skeleton003, 2025-06-26:
Currently there is `ros2_interface_library()` that can convert `[".action", ".msg", ".srv"]` to C++, python, and Rust code(, which is an amazing feature).

The underlying process is `*.msg --> *.idl --> *.h`. I wonder if it is possible to provide an API that can take `*.idl` files as input and create cc_library or py_library? For example:

```
idl_interface_library(
    name = "my_msgs",
    srcs = glob(["my_msgs/msg/*.idl"]),
    visibility = ["//visibility:public"],
    deps = [...],
)

cpp_idl_interface_library(
    name = "cpp_my_msgs",
    visibility = ["//visibility:public"],
    deps = [":my_msgs"],
)
```

mvukov, 2025-06-28:
Hi, we just need to add support in `ros2_interface_library` to handle .idl files as well. Everything else stays the same. Out of curiosity: why do you use .idl files and not .msg files?

Skeleton003, 2025-07-02:
> Hi, we just need to add support in `ros2_interface_library` to handle .idl files as well. Everything else stays the same. Out of curiosity: why do you use .idl files and not .msg files?

It was just an attempt out of curiosity, and now I've given it up. But still thank you!

closed 2025-07-02 by Skeleton003